stm32-rs/stm32f0xx-hal #8: [META] Clarification needed on adding additional devices

jessebraham, 2018-12-18:
I'm looking to start adding support for a couple of devices (`STM32F070` and `STM32F091`), and I'm just looking for some clarification on something before I get started. I'm still relatively new to Rust so hopefully I'm not missing something simple :smile:

Initial support for the `STM32F030`, which is quite similar to the `STM32F070`, was recently added to the repository, so I planned on basing my patch(es) off of the work already done.  Most of the changes made so far make sense to me; at this time the major point of confusion for me lies in `Cargo.toml`.

This all makes perfect sense, and is exactly how I would expect it to look:
```toml
[features]
rt = ["stm32f0/rt"]
stm32f042 = ["stm32f0/stm32f0x2"]
stm32f030 = ["stm32f0/stm32f0x0"]
```

The remainder, however, has me a bit confused:
```toml
stm32f030x4 = ["stm32f030x6"]
stm32f030x6 = ["stm32f030"]
stm32f030x8 = ["stm32f030", "stm32f0/stm32f0x0"]
stm32f030xc = ["stm32f030", "stm32f0/stm32f0x0"]
```

My first question is, why does the `STM32F030` have a feature for the base model as well *as the four available variants*, while the `STM32F042` only has the base model listed as a feature, *ignoring its variants*?

Second, I'm not quite understanding the dependencies listed for the `STM32F030` variants; I'm not sure if an error has been made here or if I'm just failing to understand it. In my mind the `stm32f030 = ["stm32f0/stm32f0x0"]` is the only line that should be required, and the remainders seems redundant, so hopefully somebody call fill in the blanks on what's going on here.

Thanks in advance, keep up the great work!

therealprof, 2018-12-18:
@jessebraham The reason for that is that STM32F030 comes in different packages which offers different peripherals while most other models have the same peripherals but some are simply not accessible externally when using a smaller package, e.g. the STM32F042.

The STM32F070 is indeed quite similar to the F030, including the peripheral nonsense, so it will also need the packages stuff, while the STM32F091 seems to be on the non-nonsense side.

jessebraham, 2018-12-18:
Perfect, that is along the lines of what I was thinking, just wanted to be certain before doing too much work. Thanks for the quick reply!

closed 2018-12-18 by jessebraham