stm32-rs/stm32f0xx-hal #11: Implement ADC functionality

HarkonenBade, 2018-12-19:
Add functionality for reading ADC values.

HarkonenBade, 2018-12-19:
I'm planning to make a go at this at some point soon, though no idea how successful i'll be.

therealprof, 2018-12-19:
ADC is very useful to read analog sensors (or the operating voltage of the MCU). I tend to prefer digital sensors where possible or binarised signals using hardware but for newcomers from the Arduino world this is a must have.

HarkonenBade, 2018-12-19:
Got some first tentative steps https://github.com/HarkonenBade/stm32f0xx-hal/tree/adc-test, gonna throw them at some hardware tomorrow and see if it even remotely works.

therealprof, 2018-12-19:
NB: I do have the hardware to generate accurate signals and measure them.

HarkonenBade, 2018-12-19:
Noted, I think I can also get hold of some of that, though for now my hopes are around 'any reading' possibly climbing as high as 'a small number when V = 0v and a large number when V = 3v3'

HarkonenBade, 2018-12-20:
Gonna work on actually implementing all the channels. Might also add some bits for getting the vref and temp if i'm feeling especially fancy.

therealprof, 2018-12-20:
> Might also add some bits for getting the vref and temp if i'm feeling especially fancy.

Those are actually quite useful because they make decent out-of-the-box examples.