stm32-rs/stm32f0xx-hal #15: Side effects from using *const addresses

therealprof, 2018-12-21:
I just noticed that using `*const` means we can't share peripherals between interrupt/exception handlers and main anymore:
```
error[E0277]: `*const stm32f0::stm32f0x2::USART2` cannot be sent between threads safely
  --> examples/i2c_hal_ssd1306alphabeter.rs:23:1
   |
23 | / static PANIC_SERIAL: Mutex<RefCell<Option<hal::serial::Tx<USART2>>>> =
24 | |     Mutex::new(RefCell::new(None));
   | |___________________________________^ `*const stm32f0::stm32f0x2::USART2` cannot be sent between threads safely
```

cc @HarkonenBade @david-sawatzke

david-sawatzke, 2018-12-22:
`Pin` (of course) also doesn't implement Sync

david-sawatzke, 2018-12-22:
We should implement Sync for Rx, Tx and Pin. They don't have rmw cycles, so they are safe from race conditions. I don't see any reason to have Send for Rx & Tx, but it's useful for Pin.

therealprof, 2019-01-13:
This has been addressed.

closed 2019-01-13 by therealprof