stm32-rs/stm32f0xx-hal #59: Wrong I2C pin definitions for stm32f030.

matoushybl, 2019-05-09:
Stm32f030 datasheet defines pinsPA11 and PA12 alternate function 5 as "SCL" and "SDA" instead of "I2Cx_SCL" and "I2Cx_SDA". Current implementation assumes that these pins are used by the I2C1 interface. This is however seems to not be true for the STM32F030 as the I2C1 interface doesn't work on these pins.

Does anyone have any further information on this?

I may try using these pins with I2C2 in the future and then open a PR with the fix.

therealprof, 2019-05-09:
Not sure what you're trying to say but CubeMX agrees that I2C is not available on PA11/PA12 so there might be a bug in the HAL impl. Interestingly CubeMX is also buggy when crosschecking with a STM32F042...

matoushybl, 2019-05-09:
I've looked into datasheets of STM32F042 and STM32F048 and these claim that I2C1 is available on these pins. 

As of now, I'd push a PR with the fix removing STM32F030 from the cfg for these pins. What do you think?

therealprof, 2019-05-09:
> I've looked into datasheets of STM32F042 and STM32F048 and these claim that I2C1 is available on these pins.

Yes, but CubeMX doesn't have it on the first footprint I happened to use for crosschecking. ;)

> As of now, I'd push a PR with the fix removing STM32F030 from the cfg for these pins. What do you think?

Yes please.

matoushybl, 2019-05-10:
PR submitted and merged.

closed 2019-05-10 by matoushybl

therealprof, 2019-05-10:
Pro tip: If you put something like Fixes #59 in your PR, the issue will be automatically closed when the PR is merged.