stm32-rs/stm32f0xx-hal #61: Flags are never cleared when there's a UART error

thejpster, 2019-05-29:
If there's a UART error (for example, it receives a byte when there's already a byte in the Receive Data Register), a sticky flag gets set. This flag causes `read()` to return an error, but the flag doesn't get cleared. So every `read()` also returns an error.

I propose you clear the flag by writing to the appropriate bit in `icf` just before the error is returned.

therealprof, 2019-06-05:
@thejpster Any chance you could test the PR? Having troubles creating errors here... ;)

thejpster, 2019-06-07:
The easiest way to generate an error is to write two characters to the UART, without reading the first one first. This will trigger a UART data overflow error.

commit 0c13812c (2019-11-04): Clear error flags before returning

Closes #61
commit 143f6b2b (2019-11-04): Clear error flags before returning

Closes #61
PR #63 (2019-11-04): Clear error flags before returning
Closes #61
closed 2019-10-06 by therealprof