stm32-rs/stm32f0xx-hal #65: 'Blink' example not working

jack6000, 2019-06-30:
The "blink" project in the example directory is not working. It can compile fine, but don't run on my nucleo F091RC board.
The led is supposed to blink, instead the led light up and that's all.
Tried to debug step by step, but can't go after the critical section call.

Btw, is there an example to blink a led without using a critical section ?

therealprof, 2019-06-30:
examples/blinky_delay.rs?

jack6000, 2019-07-02:
No, the examples/blink.rs

Sorry, but it was my fault. i forgot that the 'Cargo build' command produce binary in debug mode so it is much slower than in release mode.
Add to this the 1 to 1_000_000 loop delay and the led would blink in an hour...

Don't know why, but the 'next' command in gdb freeze everything, while single stepping is working.

Thanks.

closed 2019-10-06 by therealprof