stm32-rs/stm32f0xx-hal #66: Unable to compile examples

Maldus512, 2019-08-01:
Hello,
I'm trying to run the blinky-delay example on an stm32f030rc. At first I have tried copying it in a new project and adding the crate as a dependency, by removing the `crate` self reference and adding version 0.14.1 to `Cargo.toml` dependencies:

```rust
#![no_main]
#![no_std]

#[allow(unused)]
use panic_halt;

use stm32f0xx_hal as hal;

use hal::{delay::Delay, prelude::*, stm32};

use cortex_m::peripheral::Peripherals;
use cortex_m_rt::entry;

#[entry]
fn main() -> ! {
    if let (Some(mut p), Some(cp)) = (stm32::Peripherals::take(), Peripherals::take()) {
        cortex_m::interrupt::free(move |cs| {
            let mut rcc = p.RCC.configure().sysclk(8.mhz()).freeze(&mut p.FLASH);

            let gpioa = p.GPIOA.split(&mut rcc);

            // (Re-)configure PA1 as output
            let mut led = gpioa.pa1.into_push_pull_output(cs);

            // Get delay provider
            let mut delay = Delay::new(cp.SYST, &rcc);

            loop {
                led.toggle();
                delay.delay_ms(1_000_u16);
            }
        });
    }

    loop {
        continue;
    }
}
```

Unfortunately this leads to the compiler complaining about unresolved imports for all modules used:
```
   Updating crates.io index
   Compiling stm32f0 v0.7.1
   Compiling stm32f0xx-hal v0.14.1
   Compiling app v0.1.0 (/home/maldus/Projects/app)
error[E0432]: unresolved imports `hal::delay`, `hal::prelude`, `hal::stm32`
  --> src/main.rs:10:11
   |
10 | use hal::{delay::Delay, prelude::*, stm32};
   |           ^^^^^         ^^^^^^^     ^^^^^ no `stm32` in the root
   |           |             |
   |           |             could not find `prelude` in `hal`
   |           could not find `delay` in `hal`

warning: unused import: `prelude::*`
  --> src/main.rs:10:25
   |
10 | use hal::{delay::Delay, prelude::*, stm32};
   |                         ^^^^^^^^^^
   |
   = note: #[warn(unused_imports)] on by default

error[E0599]: no method named `mhz` found for type `{integer}` in the current scope
  --> src/main.rs:19:54
   |
19 |             let mut rcc = p.RCC.configure().sysclk(8.mhz()).freeze(&mut p.FLASH);
   |                                                      ^^^

error: aborting due to 2 previous errors

Some errors have detailed explanations: E0432, E0599.
For more information about an error, try `rustc --explain E0432`.
error: Could not compile `app`.
```

I have since tried to compile examples directly from the repository itself, but I encountered similar issues. I'm working on stable, do I need a particular toolchain version to compile this?
```
$ rustc -V
rustc 1.36.0 (a53f9df32 2019-07-03)
```

therealprof, 2019-08-01:
Have you added the `stm32f030xc` feature to your stm32f0xx-hal reference in Cargo.toml?

Maldus512, 2019-08-02:
No, I didn't. Now it works, thank you.
Maybe it should be added to the README

therealprof, 2019-08-02:
Please feel free to send a PR to add relevant information to the README.

closed 2019-08-03 by Maldus512