stm32-rs/stm32f0xx-hal #68: Clarification on Embedded Rust programming philosophy

Maldus512, 2019-08-03:
Hello,
I'm starting a new project with an stm32f0 device which I specifically chose to be able to develop Rust firmware. I made a few experiments with just registers and the `cortex_m` crate following along the Embedded Rust book's teaching examples to really understand how the architecture works, then started approaching this library for ease of use.

I have a few years of experience as an embedded C developer, a few months of tutorials and experiments with Rust and overall good comprehension of modern programming language features. I'm having a few difficulties in understanding what is the proper way to do things in Rust and I could use some clarifications.

I took the blinky example and started working from that: I wanted to add a systick interrupt to keep track of milliseconds elapsed since the start:
```rust
#[exception]
fn SysTick() -> ! {
    unsafe {
        MILLIS += 1;
    }
}
```
and then read the shared variable `MILLIS` to manage various temporized functions in the main cycle. When I tried this approach however I realized that it didn't work because the main cycle (following the blinky example) was wrapped in a `CriticalSection` context, disabling all interrupts by default.
When I tried moving the operations outside of the critical section (i.e. blinking the LED every 1000 milliseconds) the compiler meticulously pointed out that I had no way of accessing the `PA0<Output<PushPull>>` reference I was using to toggle the LED. Plus, I can't even create one without a critical section token.

In fact, there is no example in the repository that has concurrent main task and interrupts: it's either just the main cycle or just the interrupt, with the main cycle stuck in an empty loop (thinking about `led_hal_button_irq.rs`). So, my first question is: how would I proceed to obtain a main cycle that allows interruptions but has obtained a reference to the peripherals? The `interrupt::free` context moves the values so I can't use them afterwards; I know I can populate a `Mutex` or `RefCell`, but those are data structures used to share and sync references between threads, and I have no need for sharing: only my main thread will access those, with the interrupts occasionally setting flags and such.

While I was looking for a solution I also dropped the idea of keeping track of time via interrupt and instead I'm using a free counter. Nonetheless I expect to hit this obstacle again soon enough.
This is the code I've written so far, just to explain what I was trying to accomplish.

```rust
#![no_main]
#![no_std]

extern crate panic_semihosting;

use stm32f0xx_hal as hal;

use hal::{prelude::*, stm32};

use cast;
use cortex_m::peripheral::Peripherals;
use cortex_m_rt::entry;
use cortex_m_rt::exception;

#[entry]
fn main() -> ! {
    let mut p = stm32::Peripherals::take().unwrap();
    let cp = Peripherals::take().unwrap();

    cortex_m::interrupt::free(move |cs| {
        p.RCC.apb1enr.modify(|_, w| w.tim6en().set_bit());
        p.RCC.apb1rstr.modify(|_, w| w.tim6rst().set_bit());
        p.RCC.apb1rstr.modify(|_, w| w.tim6rst().clear_bit());
        let mut rcc = p.RCC.configure().sysclk(8.mhz()).freeze(&mut p.FLASH);

        let gpioa = p.GPIOA.split(&mut rcc);

        // (Re-)configure PA1 as output
        let mut led = gpioa.pa12.into_push_pull_output(cs);
        let tim = p.TIM6;

        // pause
        tim.cr1.modify(|_, w| w.cen().clear_bit());
        // restart counter
        tim.cnt.reset();

        let frequency = 1000;
        let ticks = rcc.clocks.pclk().0 / frequency;

        let psc = cast::u16(ticks - 1).unwrap(); // / (1 << 16)).unwrap();
        tim.psc.write(|w| unsafe { w.psc().bits(psc) });

        // start counter
        tim.cr1.modify(|_, w| w.cen().set_bit());

        let mut ts: u32 = 0;

        /* The main loop should be outside of the interrupt-free context */
        loop {
            if is_expired(ts, get_millis(&tim), 1000) {
                led.toggle();
                ts = get_millis(&tim);
            }
            check_overflow(&tim);
        }
    });

    loop {
        continue;
    }
}

static mut MILLIS: u32 = 0;

pub fn is_expired(start : u32, current : u32, delay : u32) -> bool {
    let finish = start.wrapping_add(delay);
    (finish.wrapping_sub(current) as i32) < 0
}

pub fn check_overflow(timer: &hal::stm32::TIM6) {
    cortex_m::interrupt::free(|_| {
        if timer.sr.read().uif().is_update_pending() {
            unsafe { MILLIS += 0xFFFF }
            timer.sr.write(|w| w.uif().clear());
        }
    });
}

pub fn get_millis(timer: &hal::stm32::TIM6) -> u32 {
    let mut result: u32 = 0;
    cortex_m::interrupt::free(|_| unsafe { result = MILLIS + timer.cnt.read().bits() });
    result
}
```

Please be patient if I'm missing a key aspect of Rust programming, I'm new to it and I promise I did my research before asking.

therealprof, 2019-08-03:
What you determined is exactly right, the CriticalSection is main is supposed to be used only for the duration of initialisation so safely setup shared resources and prevent interrupts from running prematurely.

If you have a look at the `flash_systick.rs` you can see how this would work.

therealprof, 2019-08-03:
I've added an improved example which transfers the control over some protected peripheral into the exception handler so it can be used without further CriticalSections: https://github.com/stm32-rs/stm32f0xx-hal/blob/master/examples/flash_systick_fancier.rs

Maldus512, 2019-08-03:
Thank you for the confirmation. Both of those examples however use data sharing structures to access the peripherals outside of the initial critical section used for initialization; for what I have in mind I don't need to share them between the main cycle and the interrupt routine: it would be enough to access the led outside the critical section but still inside the main function.

Is the preferred way to use Mutexes and RefCells even in this scenario or do I have less convoluted options? I guess this question is more about Rust in general than its embedded side, since I'm basically asking how to access a reference after it has been moved to a closure.

therealprof, 2019-08-04:
>  it would be enough to access the led outside the critical section but still inside the main function.

No problem, just use them in the main function then. I've updated https://github.com/stm32-rs/stm32f0xx-hal/blob/master/examples/blinky.rs to show how to limit the duration of the CS and move initialised peripherals out of it for use in the `main` function.

Maldus512, 2019-08-04:
Many thanks again. I didn't realize I could return values from the closure used in `free`, now everything is completely clear.

closed 2019-08-04 by Maldus512

therealprof, 2019-08-04:
@Maldus512 Please feel free to add some text about your stumbling blocks and resolutions to the available documentation so other users can profit.