stm32-rs/stm32f0xx-hal #71: I2C, SPI, and other peripherals should require their respective DMA channels

bobtwinkles, 2019-10-06:
Currently, it's possible to configure the SPI1 and I2C1 peripherals like this:

```
            let sda = gpiob.pb7.into_alternate_af1(cs);
            let scl = gpiob.pb6.into_alternate_af1(cs);

            let i2c = hal::i2c::I2c::i2c1(p.I2C1, (scl, sda), 100.khz(), &mut rcc);

            let sck = gpioa.pa5.into_alternate_af0(cs);
            let miso = gpioa.pa6.into_alternate_af0(cs);
            let mosi = gpioa.pa7.into_alternate_af0(cs);

            let spi = hal::spi::Spi::spi1(
                p.SPI1,
                (sck, miso, mosi),
                ws2812_spi::MODE,
                3_000.khz(),
                &mut rcc,
            );
```

However, this configuration doesn't actually work since on most STM32F0xx parts, `SPI1` and `I2C1` share DMA channels 2 (TX) and 3 (RX). Even if you don't use the DMA features, the attempt to share resources puts both peripherals into a bad state. On `STM32F07x` parts we could use the `I2C1_DMA_RMP` or `SPI1_DMA_RMP` bits in `SYSCFG_CFGR1` to remap one of the two peripherals to use channels 6&7, so any API for this will need to support that somehow.

This also raises a related concern, which is that there doesn't seem to be a safe way to disable peripherals. My expectation would be that the `release()` methods would disable them, but looking at the implementations that doesn't seem to be the case. To support it, we'd probably have to pass the `RCC` into `release()`.

therealprof, 2019-10-06:
>  Even if you don't use the DMA features, the attempt to share resources puts both peripherals into a bad state. 

Is that so? I've done it before without problems and CubeMX also doesn't complain about that combination.

> This also raises a related concern, which is that there doesn't seem to be a safe way to disable peripherals. My expectation would be that the release() methods would disable them, but looking at [...]

I agree. We could also do it using a criticial section instead of taking a RCC parameter.

bobtwinkles, 2019-10-06:
> Is that so? I've done it before without problems and CubeMX also doesn't complain about that combination.

On an STM32F042k6, if I try to enable both peripherals using the code above the I2C peripheral gets stuck waiting for the TXIs flag to go low. Maybe it's only on these smaller parts that there is an issue? I don't have a logic analyzer (yet) so I don't know what the bus is actually doing... I'll try to track that down sometime next week when it arrives. 

> I agree. We could also do it using a criticial section instead of taking a RCC parameter.

That feels a little heavyweight -- what should compile down to a single register read-modify-write would require enabling/disabling interrupts. I don't know how expensive that is on the STM32s though, so maybe it's fine.

therealprof, 2019-10-07:
> On an STM32F042k6, if I try to enable both peripherals using the code above the I2C peripheral gets stuck waiting for the TXIs flag to go low. Maybe it's only on these smaller parts that there is an issue?

The STM32F042 is one of my goto devices. Haven't seen the problem there. I can try to specifically reproduce with your suggested configuration the next couple of days. I also do have oscilloscope and logic analyser if needed.

> That feels a little heavyweight -- what should compile down to a single register read-modify-write would require enabling/disabling interrupts. I don't know how expensive that is on the STM32s though, so maybe it's fine.

Costs a few cycles, no biggie. I wouldn't expect someone to release peripherals frequently and in performance sensitive code anyway.

bobtwinkles, 2019-10-10:
I had a chance to play with this some more this evening, and it seems like the issue is related to conflicting pin assignments. If I configure only pin `pa7` to AF0 (and don't start the SPI peripheral), I2C keeps working. Configuring any of the other pins used above (PA5 or PA6) to AF0 breaks the I2C peripheral.

therealprof, 2019-10-10:
I just checked https://www.st.com/resource/en/errata_sheet/dm00108473.pdf and couldn't find anything related. Still haven't had time to look at the hardware yet.

therealprof, 2019-10-10:
Okay, just tried it on the STM32F072DISCO, no such problem. Did you forget the pull-up maybe and this really is an electrical problem? (Just curious because I always use the internal pull-ups but your example code doesn't).

therealprof, 2019-10-10:
FWIW I've added a working I2C example (without the SPI pin configuration, but it's easy to copy them in from above) here: https://github.com/stm32-rs/stm32f072b-disco/blob/master/examples/i2c_hal_ssd1306alphabeter.rs

bobtwinkles, 2019-10-12:
I was able to get things working using the following pin configuration:

```
             let sck = gpiob.pb3.into_alternate_af0(cs);
             let miso = gpiob.pb4.into_alternate_af0(cs);
             let mosi = gpiob.pb5.into_alternate_af0(cs);
 
             let sda = gpiob.pb7.into_alternate_af1(cs);
             let scl = gpiob.pb6.into_alternate_af1(cs);
```

This is the configuration recommended by ST for SPI peripheral on NUCLEO-F042K6 boards.

I don't /think/ it's an electrical problem, since:
   - The I2C bus works fine when the SPI peripheral isn't configured
   - There are external pullups on the SDA and SCL lines.

Looking at the output from a logic analyzer, it seems like the SPI data is showing up on I2C1 SDA (i.e. PB7).

therealprof, 2019-10-12:
Well, don't know what to tell you or do about it. Neither my own tests nor the documentation nor the datasheets nor the errata agree with your discoveries.

I guess this one of the "Doctor, doctor, it hurts when I do this..." cases.

bobtwinkles, 2019-10-12:
That does seem to be the case =). Playing with it some more, it seems like attaching SPI1 to anything on GPIOA causes the whole GPIO frontend to fall over hard enough that it can't even wiggle pins? I don't have the tools to really debug this further, so I think I'll just close this out.

closed 2019-10-12 by bobtwinkles