stm32-rs/stm32f0xx-hal #86: USB driver release

michaelbeaumont, 2020-01-18:
In the current [`CHANGELOG`](https://github.com/stm32-rs/stm32f0xx-hal/blob/02278e554201c7b1a3dcf4bd121b4306b7bbf04e/CHANGELOG.md), we see `Added USB driver` under `0.15.2`, but this feature is missing from that version. As far as I can tell, no released version contains the `stm32-usbd` feature.

Seeing as the newest release of `stm32-usbd` no longer has the board-specific `impl`s, I think it'd be good to release a version of this crate with the aforementioned feature.

Thanks!

pigrew, 2020-01-31:
I think that this should be resolved with the next release of this stm32f0xx-hal crate?

The `stm32-usbd-examples` seems to not have been updated after the USB driver's feature flags were revised.... but that's an issue with that repository.

I've been making many changes to the RCC module in order to get it working on the stm32f070rb (which I just did). With the master branch of the `HAL`, on this MCU, I had to change the clock configuration (to use HSE and set the USB clksrc), and also set the LED pin definition. At some point, I'll update the usb-serial example (in this repository) so that it's more obvious how to get it working on a wider range of targets.

michaelbeaumont, 2020-01-31:
Yeah sure, the next release resolves it, I just thought maybe a quick fix release could be made with that feature/I wanted to document the situation for others so they don't waste time.
Seemed to me that it was an oversight where everyone thought it had been released, since the change was added to the `CHANGELOG` for `0.15.2` by mistake.
And it's been so long since the change to `stm32-usbd` was made, predicated on the board crates updating, and others have released USB support.

But no problem for me, I'm just linking to the `master` now

therealprof, 2020-01-31:
Yeah, sorry about missing that. I've moved it to the correct place now but we probably should do a new release rather sooner than later anyway.

commit c39da7e3 (2020-01-31): Moved USB driver to unreleased release

Closes #86
closed 2020-01-31 by therealprof