stm32-rs/stm32f0xx-hal #89: Conflicting types for 'assert_failed'

david-oz00, 2020-01-30:
When USE_FULL_ASSERT is enabled, STM32CubeMX generates this function in _main.c_:

`void assert_failed(char *file, uint32_t line)`

however the prototype in _stm32f0xx_hal_conf.h_ is:

`void assert_failed(uint8_t* file, uint32_t line);`

This causes a compile error from GCC "error: conflicting types for 'assert_failed'".  It looks like someone updated the prototype but forgot to update the body.  The funny thing is _**uint8_t***_ is not correct anyway.  I think it should be changed to _**const char***_, or at least keep it at _**char***_

therealprof, 2020-01-30:
I think you picked the wrong GitHub project for your issue report. This is all about support in the **Rust** programming language, there's no C code.

closed 2020-01-30 by therealprof

david-oz00, 2020-01-30:
Sorry, my bad.  Should have paid more attention.

therealprof, 2020-01-30:
No worries, maybe we'll see you on the other side some time. ;)