stm32-rs/stm32f0xx-hal #90: Timer<TIM7> fires twice as fast as expected

jamesmunns, 2020-02-01:
Hey, using the STM32F072, a periodic 1khz timer seems to actually be running at 2khz.

My RCC looks like this:

```rust
let mut rcc = rcc
    .configure()
    .hsi48()
    .enable_crs(crs)
    .sysclk(48.mhz())
    .pclk(24.mhz())
    .freeze(&mut flash);
```

My Timer setup looks like this:

```rust
let mut ms_timer = Timer::tim7(tim7, 1.khz(), &mut rcc);
ms_timer.listen(Event::TimeOut);
```

And my timer interrupt looks like this:

```rust
#[task(binds = TIM7, resources = [led, ms_timer])]
fn clock(cx: clock::Context) {
    static mut TOGGLE: bool = false;
    static mut STEPDOWN: usize = 0;

    // Only way to clear the interrupt flag
    cx.resources.ms_timer.wait().ok();

    *STEPDOWN += 1;

    if *STEPDOWN >= 1000 {
        *STEPDOWN = 0;
    } else {
        return;
    }

    if *TOGGLE {
        cx.resources.led.set_low().ok();
    } else {
        cx.resources.led.set_high().ok();
    }
    *TOGGLE = !*TOGGLE
}
```

With this, I would expect that the LED blinks at approximately 0.5Hz (it toggles at 1Hz), however it seems to be blinking at 1Hz (toggling at 2Hz). I'm not sure where the timer clock source is set (the code calculation uses pclk(), the reference manual mentions "CK_CNT", which I can't find in the HAL code. I have a suspicion that it is using the core clock (at 48MHz), but using the peripheral clock (at 24MHz) to calculate the counter value.

jamesmunns, 2020-02-01:
Specifically I am using git commit b15fc1532f738e06df23935446ce4bfa67b078cc, which is after #87 has landed. Not sure if this behavior is different before/after this patch.

jamesmunns, 2020-02-01:
I tried backing out #87 manually, and it still blinks twice as fast as expected.

therealprof, 2020-02-02:
@jamesmunns The problem is something trivial. For some reason the frequency fed into the timers is automatically doubled if PCLK is prescaled which is exactly what you're requesting in your clock configuration.

PR #91 (2020-02-02): Assume twice the timer frequency when HCLK != PCLK
Internally the frequency going to the timers is doubled if PCLK is
prescaled.

Fixes #90
commit 53e6f731 (2020-02-02): Assume twice the timer frequency when HCLK != PCLK

Internally the frequency going to the timers is doubled if PCLK is
prescaled.

Fixes #90
closed 2020-02-02 by therealprof