stm32-rs/stm32f0xx-hal #97: Loop {} in main() seems to prevent interrupts from firing.

nebelgrau77, 2020-02-20:
Not sure if that's an issue or something is missing in the interrupts setup: 

I tried using the blinky-irq example to do something else, e.g. display something on SSD1306, while the interrupts are updating values, toggling LEDs, etc. 
As it didn't work, I tried adding single pieces one by one. The conclusion is: as soon as I put a loop {} in main () in this code https://github.com/nebelgrau77/STM32F0_blinky_IRQ 
the LEDs stop blinking. It doesn't matter if it's an empty loop, or with a NOP, or with any actual stuff being done.

therealprof, 2020-02-20:
Where do you put the `loop`? is that linked code the code which is not working?

nebelgrau77, 2020-02-20:
The code is working as linked. If I put a loop between 142 and 143, it won't.

therealprof, 2020-02-20:
Ah, now I see your problem, you can't do that since you're in a critical section, so interrupts are disabled. If you put an endless loop into the critical section interrupts will never fire. If you want to do something in the main loop you need to do it e.g. where the current `loop { continue; }` resides.

closed 2020-02-20 by therealprof

nebelgrau77, 2020-02-20:
Oh, gotcha... So for my display example I would somehow make it globally available after setup, and have it updated in that external loop. Thanks!

therealprof, 2020-02-20:
Yes, you can return the components you need out of the critical section closure.

nebelgrau77, 2020-02-20:
OK, but at this point I have two questions, if you don't mind sheding some light on this:
1) Do I actually need to do the setup inside the critical section? Looking at some examples for F1 and F4 I see it done like this: 

```
 if let (Some(dp), Some(_cp)) = (
        stm32::Peripherals::take(),
        cortex_m::peripheral::Peripherals::take(),
    ) { //peripherials setup happens here; 

```
Is it something specific to the way the F0 crate works, or it has some other advantages?

2) If I want to make the SSD1306 global, which part has to be made global: SCL and SDA pins, I2C, the SSD1306 instance itself? 

Thanks!

therealprof, 2020-02-20:
Well, you need the CS because the code doing the GPIO initialisation is doing read-modify-write operations which cannot be guaranteed on Cortex-M0. In general I would consider it good practise to at least run the interrupt initialisation code with interrupts disabled. But feel free to move anything out of the big CS or only have dedicated short CSes whenever they're needed (which you can spot by having the cs argument).

Regarding I2C, it depends: I'd probably initialise as much as possible and hand the `display` instance around. I do have some F0 code for the SSD1306 here: https://github.com/stm32-rs/nucleo-f042k6/tree/master/examples and also here: https://github.com/therealprof/MCUmeter-software/blob/master/src/main.rs

nebelgrau77, 2020-02-20:
OK, gotcha. I got the SSD1306 on F0 figured out pretty well by now, but I need to figure out how to take it outside :) Thanks!!!