stm32-rs/stm32f0xx-hal #104: 0.16.0 docs build is failing

newAM, 2020-04-05:
docs.rs failed to build 0.16.0: https://docs.rs/crate/stm32f0xx-hal/0.16.0

[logs](https://docs.rs/crate/stm32f0xx-hal/0.16.0/builds/217861)
```
[INFO] [stderr] WARNING: Your kernel does not support swap limit capabilities or the cgroup is not mounted. Memory limited without swap.
[INFO] [stdout] add606e904d8d00e9e83ceb3d5cf36d06acbc0cfc0cd35183e9f98ced61ac806
[INFO] running `"docker" "start" "-a" "add606e904d8d00e9e83ceb3d5cf36d06acbc0cfc0cd35183e9f98ced61ac806"`
[INFO] [stderr]    Compiling semver-parser v0.7.0
[INFO] [stderr]    Compiling typenum v1.11.2
[INFO] [stderr]    Compiling proc-macro2 v1.0.8
[INFO] [stderr]    Compiling unicode-xid v0.2.0
[INFO] [stderr]    Compiling cortex-m v0.6.2
[INFO] [stderr]     Checking stable_deref_trait v1.1.1
[INFO] [stderr] error[E0463]: can't find crate for `core`
[INFO] [stderr]   |
[INFO] [stderr]   = note: the `thumbv6m-none-eabi` target may not be installed
[INFO] [stderr] 
[INFO] [stderr] error: aborting due to previous error
[INFO] [stderr] 
[INFO] [stderr] For more information about this error, try `rustc --explain E0463`.
[INFO] [stderr] error: could not compile `stable_deref_trait`.
[INFO] [stderr] warning: build failed, waiting for other jobs to finish...
[INFO] [stderr] error: build failed
[INFO] running `"docker" "inspect" "add606e904d8d00e9e83ceb3d5cf36d06acbc0cfc0cd35183e9f98ced61ac806"`
[INFO] running `"docker" "rm" "-f" "add606e904d8d00e9e83ceb3d5cf36d06acbc0cfc0cd35183e9f98ced61ac806"`
[INFO] [stdout] add606e904d8d00e9e83ceb3d5cf36d06acbc0cfc0cd35183e9f98ced61ac806
```

therealprof, 2020-04-05:
Probably a weird docs.rs problem?

newAM, 2020-04-05:
Figured it out!  Targets for docs need to be specified in `Cargo.toml` now.  [Soucre](https://blog.rust-lang.org/2020/03/15/docs-rs-opt-into-fewer-targets.html)

[docs.rs setup](https://github.com/rust-lang/docs.rs/blob/master/README.md#cli)
Local reproduction of current issue: 
`docker-compose run web build crate stm32f0xx-hal 0.16.0`
Testing local fix:
`docker-compose run -v /home/alex/git/stm32f0xx-hal/:/build web build crate --local /build`

Pull request open [here](https://github.com/stm32-rs/stm32f0xx-hal/pull/105).

PR #105 (2020-04-05): Add thumbv6m-none-eabi target to docs.
Fixes #104
closed 2020-04-05 by therealprof