stm32-rs/stm32f0xx-hal #118: Have PA9/10/11/12 USB pins automatically remap?

brainstorm, 2020-08-18:
I was going a bit nuts because my board was not USB-enumerating (for a seemingly simple RTIC example) and it turned out to be that I didn't uncomment those:

```rust
        /* Uncomment the following lines if you have a chip in TSSOP20 (STM32F042F)
        or UFQFPN28 (STM32F042G) package
        This code enables clock for SYSCFG and remaps USB pins to PA9 and PA10.
        */
        dp.RCC.apb2enr.modify(|_, w| w.syscfgen().set_bit());
        dp.SYSCFG.cfgr1.modify(|_, w| w.pa11_pa12_rmp().remapped());
```

Fortunately, @mvirkkunen and other very helpful folks at rust-embedded discord channel pointed that out...

Now, would it be at all possible to detect those different variants automatically, somehow?

mvirkkunen, 2020-08-18:
I think it might be possible by creating a different `Peripheral` struct for the PA9/PA10 pins that's available on certain devices, and having the `enable` function remap them. Or alternatively have a special constructor for it.

therealprof, 2020-08-18:
I don't see how. The chips are just bonded into different packages and are not self-aware of how they're bonded.

brainstorm, 2020-08-18:
> ...to do the remapping without having to mess with the registers by hand, that is. It's probably not possible to detect which chips have which pins, because it's package dependent.

Yes, I meant this, sorry for the confusion @therealprof, I should have phrased that better :-S

therealprof, 2020-08-18:
@brainstorm Thanks for the clarification. Yes, that makes absolute sense.

Want to take a stab at it?

brainstorm, 2020-08-18:
Sure, I don't know what I'm doing, but let's take it on PR #119 ;)

closed 2020-08-18 by brainstorm