stm32-rs/stm32f0xx-hal #121: cannot find value `RCC_PLLSRC_PREDIV1_SUPPORT`

pdgilbert, 2020-09-30:
With my own code and also with a copy of examples/blinky_delay.rs, using git master and staging, I am getting
```
Compiling stm32f0xx-hal v0.17.1 (https://github.com/stm32-rs/stm32f0xx-hal#e679ab19)
error[E0425]: cannot find value `RCC_PLLSRC_PREDIV1_SUPPORT` in module `self::inner`
   --> /../.cargo/git/checkouts/stm32f0xx-hal-3a41196e5b12fb2e/e679ab1/src/rcc.rs:457:66
    |
457 |             let pllprediv = match (&self.clock_src, self::inner::RCC_PLLSRC_PREDIV1_SUPPORT) {
    |                                                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^ not found in `self::inner`
```
Am I missing something simple, or is this a bug?

pdgilbert, 2020-09-30:
Actually, the problem is with building the crate, so same problem in all examples.

therealprof, 2020-09-30:
How are you compiling?

pdgilbert, 2020-09-30:
For the copy of blinky_delay, at home in my package directory I use
```
cargo build --target thumbv6m-none-eabi  --features "stm32f0xx, stm32f030" --example blinky_delay
```
but more detail is in the CI at https://travis-ci.org/github/pdgilbert/eg_stm_hal/jobs/731688063

therealprof, 2020-09-30:
Not sure what "stm32f0xx" is supposed to do as that feature doesn't exist but unfortunately for the F030 you will have to specify a concrete package version, e.g. `stm32f030x4` instead of just `stm32f030`. Maybe we can improve this in the crate...

closed 2020-10-01 by pdgilbert