stm32-rs/stm32f0xx-hal #139: SPI on 0.17.1 is broken

romixlab, 2021-07-11:
Absolutely the same code is working fine on 0.16 and doesn't on 0.17. I believe this is due to some performance updates merged lately. There is simply nothing on SCK or MOSI pins. Initialisation code is stanard:
```rust
const MODE: Mode = Mode {
        polarity: Polarity::IdleLow,
        phase: Phase::CaptureOnFirstTransition,
};
let spi = Spi::spi1(p.SPI1, (mcp_sck, mcp_miso, mcp_mosi), MODE, spi_freq, &mut rcc);
```
Usage is also standard, `write` and `transfer` methods are used.

[image]
[image]

therealprof, 2021-09-04:
Sorry for the late reply. Have you tried whether master addresses your problem?

romixlab, 2021-09-12:
Seems to be working fine, not sure which commit fixed it though. If you can point it out, I can double check both.

therealprof, 2021-09-13:
No worries, just want to make sure this has been adressed already, thanks! Will close this issue then.

closed 2021-09-13 by therealprof