stm32-rs/stm32f0xx-hal #162: PWM on Special Channels?

AdinAck, 2023-05-17:
Hello, I am using an STM32F031K6 and I am trying to modify the pwm example to output complementary pwm signals on PA7 and PA8 (since PA8 is on TIM1_CH1 and PA7 is on TIM1_CH1N).

I cannot figure out how to do this, when I try to add PA7 to the `channels` declaration the `pwm::tim1` function yells at me.

I have determined any pin on CH1, 2, 3, or 4 are accepted, but any on BKIN, ETR or CH1N are not.

Am I going about this the wrong way? Is there a different way to configure pins to use these special channels?

Thanks!

AdinAck, 2023-05-18:
It seems that the exact functionality I am talking about is in the f4 HAL [here](https://github.com/stm32-rs/stm32f4xx-hal/blob/master/examples/pwm-dead-time.rs).

The interface is exactly what I would expect, why does the f0 HAL not have this?

therealprof, 2023-05-18:
This HAL is very much on the backburner, especially compared to the f4 HAL where this was added actually just a few weeks ago.

AdinAck, 2023-05-18:
I see, I am very new to Rust but I will do my best to contribute.

I will leave this issue open until the desired functionality is added.

closed 2023-09-01 by AdinAck