stm32-rs/stm32f0xx-hal #183: CAN does not fulfil trait requirements to be used in bxcan

Jack17432, 2024-12-02:
```
let mut can1 = {
    let (pin_can_tx, pin_can_rx) = cortex_m::interrupt::free(|cs| {
        (
            gpioB.pb9.into_alternate_af4(cs),
            gpioB.pb8.into_alternate_af4(cs),
        )
    });

    let can_inst = CanInstance::new(ctx.device.CAN, pin_can_tx, pin_can_rx, &mut rcc);

    Can::builder(can_inst)
        .set_bit_timing(0x001a000b) // http://www.bittiming.can-wiki.info/
        .set_silent(false)
        .set_loopback(false)
        .set_automatic_retransmit(true)
        .enable()
};
```

```
error[E0277]: the trait bound `CanInstance<stm32f0xx_hal::gpio::gpiob::PB9<Alternate<stm32f0xx_hal::gpio::AF4>>, stm32f0xx_hal::gpio::gpiob::PB8<Alternate<stm32f0xx_hal::gpio::AF4>>>: bxcan::Instance` is not satisfied
   --> src/bin/acc24.rs:95:26
    |
95  |             Can::builder(can_inst)
    |             ------------ ^^^^^^^^ the trait `bxcan::Instance` is not implemented for `CanInstance<PB9<Alternate<AF4>>, PB8<Alternate<AF4>>>`
    |             |
    |             required by a bound introduced by this call
```

It seems that it can't get the `bxcan::Instance` due to probably the generic pins. What is the intended way to use bxCan with the stm32f0 hal?

TeXitoi, 2024-12-03:
I can’t reproduce, this compile:

```rust
#![no_main]
#![no_std]

use crate::hal::prelude::*;
use cortex_m_rt::entry;
use hal::can::{bxcan, CanInstance};
use panic_halt as _;
use stm32f0xx_hal as hal;

#[entry]
fn main() -> ! {
    let mut p = hal::pac::Peripherals::take().unwrap();
    let mut rcc = p
        .RCC
        .configure()
        .hsi48()
        .enable_crs(p.CRS)
        .sysclk(48.mhz())
        .pclk(24.mhz())
        .freeze(&mut p.FLASH);
    let gpiob = p.GPIOB.split(&mut rcc);

    let _can1 = {
        let (pin_can_tx, pin_can_rx) = cortex_m::interrupt::free(|cs| {
            (
                gpiob.pb9.into_alternate_af4(cs),
                gpiob.pb8.into_alternate_af4(cs),
            )
        });

        let can_inst = CanInstance::new(p.CAN, pin_can_tx, pin_can_rx, &mut rcc);

        bxcan::Can::builder(can_inst)
            .set_bit_timing(0x001a000b) // http://www.bittiming.can-wiki.info/
            .set_silent(false)
            .set_loopback(false)
            .set_automatic_retransmit(true)
            .enable()
    };

    loop {}
}
```

Jack17432, 2024-12-03:
I wasn't using the bxcan that is exposed from the Hal. Used the one in Hal::can and it worked.

closed 2024-12-03 by Jack17432