stm32-rs/stm32f0xx-hal #187: README.md - issue

duaneellissd, 2025-07-12:
So I am following the process of getting the hal examples to build.

I can execute the command:
```
      cargo build --features=stm32f030x4
```

But there is no docs for the next step in this process, ie: there are lots of "examples" - 
how do I build the specific example? ie blinky..

I am very familiar with this with GCC - for GCC - I am expecting an ELF file, or an "a.out" or simular

That next step seems to be missing in the README :-(

My ultimate goal is I need RUST for AARCH64 and RISCV...
But I am starting with this because I have one

Thanks.

newAM, 2025-07-16:
This is a good starting point for general embedded rust: https://docs.rust-embedded.org/book/intro/index.html

Examples can be built with `cargo build --features=stm32f030x4 --example $EXAMPLE_NAME_HERE` where `$EXAMPLE_NAME_HERE` matches the file name without the `.rs` extension.
There will be an ELF file under `target/thumbv6m-none-eabi/debug/` if you want the ELF file itself.

You can also install `probe-rs` and adjust the [runner settings](https://github.com/stm32-rs/stm32f0xx-hal/blob/8eda8641428e901c4987aca292a9c4c81f9ca63c/.cargo/config.toml#L2) to match your target and then do `cargo run --features=stm32f030x4 --example $EXAMPLE_NAME_HERE` to run it on-target.

duaneellissd, 2025-07-16:
>>  Examples can be built with 

```
      cargo build --features=stm32f030x4 --example $EXAMPLE_NAME_HERE

       where $EXAMPLE_NAME_HERE matches the file name without the .rs extension.
       There will be an ELF file under target/thumbv6m-none-eabi/debug/ if you want the ELF file itself.
```
that tidbit of information is what is missing from the readme.md file